JGMaker3dofficial/artistd #1: Repository Structure

mikejr83, 2021-01-27:
If I may offer some suggestions. The repo structure here is not conducive for collaborative work nor does it follow git recommendations.

First, in git you should avoid committing binary files. This is why GitHub has the "Releases" functionality where you can tag a specific point in the repository's history and add any binary files. These are kept outside of the git repository so that things like change detection and history do not need to move the complete binary file from point to point. This may not be an issue at first but as more releases happen and that file gets updated each new pull or clone of the repository will essentially be pulling every copy of every release ever made. This can be time consuming for a new developer.

Second, in order to allow firmware to best receive updates from the upstream Marlin team the repository should be forked from their repository. Forking clones the repository to your account and GitHub maintains a reference from the new repository in your account to the source repository. This makes comparing changes from the official firmware to the specific Artist-D firmwares easy. Also, and very important to OSS developers is that it retains authorship and history of the changes made. At the moment all files show only appear only modified by JGMaker and have no history aside from the initial commit. With a forked repository information the original authors made during the commit can be referenced and links to original issues within the source repository are maintained. This can be super helpful if there is a belief that something is not working correctly. A developer can go and look and see if it was changed for some reason or if that is in fact a bug.

Finally, the repository structure is not the same as the Marlin repo so it makes bring upstream changes into this repository hard. It means that code has to be copied and pasted into the files. This breaks the authorship chain (which is unethical) and makes common git tasks such as `cherry-pick` and `rebase` unavailable for use. 

Please see my [personal repository](https://github.com/mikejr83/Marlin-ArtistD) for Marlin that I have been using to base firmwares for my personal printers, including the Artist-D, as an example of what I am suggesting.

I am available to help with this project if you so desire.

attacktwinkie, 2021-01-27:
This is indeed one of my goals.  I'd like to re-fork Marlin then sprinkle in the JGMaker goodness.

mikejr83, 2021-01-27:
Bill and @JGMaker3d I talked with Mark G. about this and I think we're in an agreement that this repository should be "archived"; basically the repository should be renamed and marked as read-only. A [fork](https://docs.github.com/en/github/getting-started-with-github/github-glossary#fork) should be made of [MarlinFirmware/Marlin](https://github.com/MarlinFirmware/Marlin). There could be some discussion on if we should also fork the MarlinFirmware/Configurations repo and follow the practice of keeping the printer firmware separate from the actual Marlin firmware code. The idea would be that we could submit pull requests with features back to Marlin without including the custom configuration files that they do not include in their firmware repository. However, if we believe that there will be changes to the firmware that require specific code changes that would be incompatible with upstream Marlin then it may not be an issue. We could also handle this with a careful branching strategy so that any code we wanted to submit upstream would be segregated.

attacktwinkie, 2021-01-27:
I've moved the Binary to the Releases section.   I'm still working out some logistics with the new Fork.

bodinjm, 2021-01-27:
Thanks Mikejr83 and attacktwinkie. I will look at making proposal once it will have the MarlinFirmware/marlin as a work base.

pinchies, 2021-01-28:
Go for it guys. This repo was only setup to demo github to Bill. I agree that it could be setup properly; a direct fork from current Marlin sounds like the best way to go forward that is future looking.

mikejr83, 2021-01-28:
Cool. Ping me whenever the new repository is up and I can start submitting some PRs!

attacktwinkie, 2021-01-28:
There's a Limitation with Github where you can only "Fork" a repository once per account,  where JGMake has several Marlin Based products. Forking for Artist-D would be short sighted.  I don't have a solution for that so we'll have to brainstorm something.  Perhaps JGMaker would fork the Repo, then we can clone their "Master Fork"... I dunno, I'm tired and can't think.

Please check it out.  Let me have your feedback on it.   It's not a perfect solution, but I think it's a great foundation to work from.

Regardless,  I think this is ready to start playing with.  It has all the commits that the original firmware is based from, and all the customized stuff that JGMaker made for the artist-d has been added in a single commit at the end.  Check out the branch.  if it ends up being good then we'll make it the main branch.

pinchies, 2021-01-28:
Another option could be to have one fork, and have a branch for each model coming off that? That's what I did in my repo for example.

mikejr83, 2021-01-28:
@attacktwinkie I ran into that same issue. I had already forked Marlin for my Ender 3 and other kickstarter printer edits. I tried to fork again for a just Artist-D version. No go. :( 

I think what @pinchies hits on is good. It's what I had in mind for the pro vs. standard. The branches could be 

* jgmaker/artistd
* jgmaker/artistd-pro
* jgmaker/magic
* jgmaker/a1
* etc...

There is one drawback here. Releases become interesting. I'm a big proponent of GitHub's release functionality (software engineer and I follow a model called GitFlow). It helps track milestones in the code with tags. While tags in git have no relationship to the concept of releases the UI that GitHub uses for them would make it confusing when releasing firmwares for different printers. The latest release for one printer would be pushed down and not be as visible when another printer's firmware was subsequently released. For the uninitiated it would make it hard to come to the site and get "the latest" for their printer.

While it bums me out that we can't use a forked repo it isn't the end of the world. It puts a little more work on the developer with respect to upstream changes and maintenance. That said I don't really have a strong opinion on one way or another and would like to hear more opinions.

pinchies, 2021-01-28:
To be honest, I think that the github releases feature is nice, but github is pretty overwhelming for most people. It's quite unfriendly for newbies.

attacktwinkie, 2021-01-28:
I've seen some releases done very well. The tags work wonders for links from pages like that wiki, I think keeping the "officially released" binaries in the releases will be good for consistency, but I don't see a problem with anyone linking back here.

mikejr83, 2021-02-24:
I think we can close this now that we have the repo set up to mirror upstream Marlin.

closed 2021-02-24 by mikejr83