JGMaker3dofficial/artistd #13: [FR] (feature request title)

mikejr83, 2021-02-02:
### Description

When a tool change occurs, a `T0` or `T1` command in g-code, the extruder which becomes active **should not** return to the position of the previous extruder when operating in IDEX Auto Park mode.

### Feature Workflow

1. `T0` or `T1` is encountered in the g-code
2. Park T<old>
3. Set T<new> to the active extruder
4. Return control flow back (do not move the parked but active extruder)

#### Additional Information

Both Cura and PrusaSlicer will then move the newly active extruder to the position it is supposed to be printing which will almost never be the position of the last extruder. This prevents the extruder from being idle while heating up over the print.

attacktwinkie, 2021-02-03:
Does this still have the problems with oozing?  I think we had chatted about adding some retraction on tool change.

mikejr83, 2021-02-03:
This prevented the blobbing I was seeing earlier. I think this feature has been requested by a number of users in the FB group. Preventing the ooze "hair" might need to be a slicer fix. I did attempt using this with `TOOLCHANGE_FILAMENT_SWAP` but that lead to some really interesting results.

PR #14 (2021-02-06): Keep newly active extruder parked on tool change
### Description

Enable `TOOLCHANGE_NO_RETURN` so that a newly activated extruder on a tool change g-code command remains parked when in IDEX Auto-Park mode. This overrides default Marlin configuration.

### Benefits

Allows for the heating of the extruder after a tool change to be performed while the nozzle is still in the ooze bucket and not over the print. The immediate removal of the prior tool places the nozzle for that extruder in its ooze bucket so that it cannot ooze on the print.

### Related Issues

Addresses and closes #13
closed 2021-02-06 by attacktwinkie