JGMaker3dofficial/artistd #16: [BUG] Filament runout purges into ooze bucket brush

mikejr83, 2021-02-03:
### Bug Description

Filament purge for filament runout is in the middle of the ooze bucket brush. This basically destroys the brush and makes it useless.

### My Configurations

Noticed with stock (JGMaker original and customized)

### Steps to Reproduce

1. Trip filament sensor
2. Insert filament
3. Purge

**Expected behavior:** Park location for the tool head to be over the ooze catching part of the bucket

**Actual behavior:** Park location is over the outer portion of the brush

attacktwinkie, 2021-02-13:
I think this might do it in configuration.h

#define NOZZLE_PARK_FEATURE    //Artist-D

#if ENABLED(NOZZLE_PARK_FEATURE)
  // Specify a park position as { X, Y, Z_raise }
  #define NOZZLE_PARK_POINT { (X_MIN_POS + 5), (Y_MAX_POS - 5), 10 }
  #define NOZZLE_PARK_XY_FEEDRATE 100   // (mm/s) X and Y axes feedrate (also used for delta Z axis)
  #define NOZZLE_PARK_Z_FEEDRATE 5      // (mm/s) Z axis feedrate (not used for delta printers)
#endif

mikejr83, 2021-02-13:
I've had mine like this: https://github.com/mikejr83/Marlin-ArtistD/blob/jgmaker/artistd/Marlin/Configuration.h#L1531 but I haven't actually tested it yet.

closed 2021-03-06 by TwinkieXLII