JGMaker3dofficial/artistd #26: [BUG] Running G28 in IDEX Duplication Mode Causes E1 to Do "Bad Stuff"

mikejr83, 2021-02-12:
### Bug Description

When the printer is issued a `G28` command when in IDEX Duplication mode the E1 extruder tries to go past its limit.

### My Configurations

Appears to be in stock JGMaker firmware and in my custom firmware from 2.07.2

### Steps to Reproduce

1. In the menu for the printer select Configuration -> IDEX Mode -> Duplication. 
2. Load a print from SD card where the print homes the printer with a G28
3. Cover your ears

**Expected behavior:**

The extruders should home. I'm not quite sure what this should be in duplication or mirror mode. They should definitely know what their endstops are and shouldn't attempt to exceed them.

**Actual behavior:** 

Appears the E1 is trying to move too far to the left.

mikejr83, 2021-02-24:
Update: This appears to be an issue with ordering in the GCODE start-up for a print. I now believe this isn't a bug with JGMaker's version of the Marlin firmware but an issue with upstream Marlin. It appears that the extruders are not respecting the soft and hard limits of the machine.

bodinjm, 2021-03-07:
Still doing it on my machine, I look into the GCODE but can't find the root of the problem

mikejr83, 2021-03-07:
@bodinjm The best I can figure is there is some issue upstream, in Marlin. If you create another printer in your slicing software for duplication mode you can use this start gcode: https://gist.github.com/mikejr83/97b7684d1f507243d08094e4bd718ba2 and will not have the issue.

bodinjm, 2021-03-09:
OK thanks, I was planing to create a "duplication mode" slicer config with half the plate and your start G-Code, I will tell you then. Thanks for your help and "devotion to the cause" of this printer. We will tame the horse !!

closed 2021-04-15 by github-actions[bot]