JGMaker3dofficial/artistd #33: Display Screen does not show print time

tjwitman, 2021-02-24:
### Bug Description

On the JGMAKER ARTIST LCD screen, the actual run time of the part being printed does not update.  However, the display updates from the time the part to be printed is selected until the printer begins to print.  In other words, it displays the warm up time only.
### My Configurations

I am using 2021-2 right from this site.

### Steps to Reproduce

1. Select IDEX mode
2. Select FULL CONTROL or AUTO PARK
3. Select part to be printed from SD card

**Expected behavior:** Once part began to print, I would expect the timer, shown in 2 places on the screen, to keep updating until the part has finished being printed.
**Actual behavior:** The time updates from the beginning of heatup until the part starts to print.  The time is then frozen.
#### Additional Information
[image]
The picture is after a 20 hour print.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mikejr83, 2021-02-24:
Hey Thomas, thanks for sending the issue our way. I forgot to ask you when we talked earlier, the progress bar showed the percentage complete, correct?

Also, in your GCODE could you look for `M75`, `M76`, or a `M77`? Specifically the 76 or 77 would be the interesting one. Those codes are start, pause, and stop for the timer. 

I don't specifically see any issues in the Marlin repo regarding this issue. In 2.0.6 they did add some functionality around this (if you start heating a chamber it would trigger timer start), but nothing regarding paused timers.

One last thing. When we were talking I mentioned testing with my custom firmware. It's based on 2.0.7.2. Are you seeing the issue there? If you want to stay on official could you try going back to an older version seeing if it is still a problem?

tjwitman, 2021-02-24:
To be honest, I didn't pay attention to the progress bar.  I'll start printing something else and see if it advances.  I'll get back to you on the rest.
Thanks

tjwitman, 2021-02-24:
This is the start of the owl GCode:

;Sliced by ideaMaker 4.0.1.4802, 2021-02-22 17:10:51 UTC-0500
;Dimension: 300.000 300.000 340.000 0.400 0.400
;Plate Shape: 0
;Origin Center: 0
;Extruder Offset #1: 0.000 0.000
;Filament Diameter #1: 1.750
;Filament Diameter #2: 1.750
;Filament Compensation #1: 94.00
;Filament Compensation #2: 94.00
;Filament Density #1: 1240.00
;Filament Density #2: 1240.00
;Bounding Box: 105.567 193.488 109.142 190.247 0.000 149.900
;Model Gap: 0.000
M221 T0 S94.00
M221 T1 S94.00
M140 S60.00
M190 S60.00
M104 T0 S220.00
M104 T1 S220.00
M109 T0 S220.00
M109 T1 S220.00
T0
;***Start Start Gcode***
M605 S0 ; Turn on Full Control Mode
G28 ; home all axes
G90; Absolute mode
M500
;M420 S1
;M900 K0.09
M82
G92 E0 ; reset extruder
T1
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
G1 X155 Y10 Z0.3 F9000.0 ; move to start-line position
G1 E3 F120 ; small purge to get going
G1 X230 Z0.3 F1000.0 E15 ; draw 1st line
G92 E0
;G1 E-2.00 F3600.0; reset extruder
; G92 E0
G1 X373 F9000.0
T0
G92 E0
G1 X145 Y10 Z0.3 F9000.0 ; move to start-line position
G1 E3 F120 ; small purge to get going
G1 X70 Z0.3 F1000.0 E15 ; draw 1st line
G92 E0 ; reset extruder
G1 E-1.50 F2400.0; reset extruder
;***End Start Gcode***
M1001
;TOTAL_NUM: 23
;PRINTING_TIME: 0
;REMAINING_TIME: 72261
;LAYER:0
;Z:0.300
;HEIGHT:0.300
M106 S0
;PRINTING: NON-OBJECT
;PRINTING_ID: -1
G1 F2400 E-1.5000
;LAYER 0.300
G0 F9000 X108.515 Y120.046
G0 F600 Z0.300
G0 F9000 X108.615 Y120.157
;TYPE:SKIRT
;WIDTH:0.500
G1 F1500 E0.0000
G1 F1200 X109.078 Y119.807 E0.0362
G1 X109.623 Y119.480 E0.0758
G1 X110.250 Y119.196 E0.1188
G1 X110.845 Y119.003 E0.1578
G1 X111.232 Y118.915 E0.1825

tjwitman, 2021-02-24:
Using Notepad ++, I searched for M75,76 77 and found nothing.

tjwitman, 2021-02-25:
To follow up, the print status bar goes completely full as soon as the print starts.  So it appears to think that the warm-up time was the complete cycle.

mikejr83, 2021-02-25:
This command is interesting `M1001`: https://marlinfw.org/docs/gcode/M1001.html

Can you see why ideamaker is putting that into the gcode?

tjwitman, 2021-02-25:
Wow.  WTF?  I'll take it out of the gcode and see what happens.  Why would they do that? Sent from my REVVL 4+

TwinkieXLII, 2021-03-06:
We good to close this issue?

closed 2021-03-06 by TwinkieXLII