JGMaker3dofficial/artistd #34: [BUG] Pause is not working

claricerogue, 2021-02-25:
### Bug Description
Pause is not working. If E0 selected, freezing and may cause temperature increasing with fire hazard because thermal runawy doesn't work. If E1 is selected the park coordenates are wrong and try to park to E0 place.

### My Configurations

I use directly .bin. I don't have used the sources.

### Steps to Reproduce

1. Print anything with any extruder
2. Pause printing through LCD

Take the extruder to its place and wait for user intervention

Freeze, increasethe temperature, take the extrude to wrong place if E1 is selected.

mikejr83, 2021-02-25:
This may be related to a fix that is made in Marlin 2.0.6. The current beta firmware here is on the prior 2.0.5 version. 

For ref: https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.6

TwinkieXLII, 2021-03-06:
I just upgraded firmware to 2.0.6 based version.  Please try 2021.3 version of the firmware.

mikejr83, 2021-03-10:
I've started testing this on the latest release. 

Here are detailed steps:

1. Start a print (tested with a simple stringing test)
2. Pause the print via the LCD (waited until after two layers were complete)
3. The printer goes into pause park mode. (*note* there is a notice on Octoprint that the printer is paused)
4. After some time there is a heater timeout as expected (*note* I believe this to be a firmware function and not part of Octoprint; Octoprint shows a notification for this)
5. When the timeout occurs an odd value is displayed on the last line for current and set temperatures. In two tests the value for current temperature went to 0 while the set value changed, 139 in one and 907 in another.
6. Pressing the knob on the front of the machine to trigger heat-up causes the main notification to change, but the temperature values remain the same.

What remains to be seen is if Octoprint is somehow interferring with the pause functionality. In both of my tests Octoprint lost connectivity with my browser. In the second one the server became unresponsive. It could be a coincidence that the server crashed at the same time but a further test with Octoprint removed from the equation is necessary.

TwinkieXLII, 2021-03-10:
There is enhanced functionality in this version for octoprint, host
actions. So it may be the firmware notifying octoprint that there's a
timeout. I know octoprint has some timeout capability too.

It would be very valuable data to have

mikejr83, 2021-03-10:
Third test completed. No Octoprint connection this time.

Steps 1-3 are the same.

4. Printer informs user that it is purging filament
5. After purge is complete the printer presents a resume menu.

I didn't wait for a timeout to occur here.

At this point I'm willing to assign this blame to Octoprint. More data by other users would be helpful. Until then I am in favor of lowering the priority of this issue.

TwinkieXLII, 2021-03-10:
>    1. Printer informs user that it is purging filament
>    2. After purge is complete the printer presents a resume menu.

Thanks for the update.

closed 2021-04-16 by github-actions[bot]