JGMaker3dofficial/artistd #35: Gibberish when setting ESteps

tjwitman, 2021-03-06:
### Bug Description

<After flashing firmware and begin setting ESteps, I select ESteps for E1 or E2, I see gibberish where the numbers are supposed to be>

### My Configurations

My zip files would be exactly what is on this page to go along with the robin.pro file

### Steps to Reproduce

<go to the Advanced page, scroll down to ESteps and select E1 or E2>

**Expected behavior:** I would expect to see the number of steps

**Actual behavior:** I actually see the familiar gibberish
[image]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mikejr83, 2021-03-07:
Looks like it's trying to display too much on the single line.

What version are you on? Just double checking to see if you're on the recently released 2.0.6 version.

tjwitman, 2021-03-07:
It's Mark's Beta that he just put up.  2021.3-beta

tjwitman, 2021-03-07:
Damn, didn't want to close it!

tjwitman, 2021-03-07:
reopened

bodinjm, 2021-03-07:
It does the same for me even in french

nlnpereira, 2021-03-08:
Same gibberish issue on E-Step configuration screen with mine.

inspectionsbybob, 2021-03-10:
Same Gibberish on 2021.3.0.6.

mikejr83, 2021-03-10:
Confirmed fixed in [v2021.4-beta ](https://github.com/JGMaker3dofficial/artistd/releases/tag/v2021.4-beta)

closed 2021-03-10 by mikejr83