JGMaker3dofficial/artistd #36: [BUG] Bed Leveling doesn't check sensors each time....

inspectionsbybob, 2021-03-06:
Just Flashed your new code onto my Artist-D.

Mine is one of the Early 100, so maybe it has more mechanical issues BUT....

Bed Leveling.... I know this is a hot topic anyway....

Corner Leveling....
Starts in the Middle, good.  
Maybe my assumptions are wrong, but I thought every time you started the corner routine it would re-zero using the Z-Switch to re-set "HOME"

I would try to adjust the bed and nothing was changing... I would make LARGE adjustments to the Zero screw, and NOTHING CHANGED.  Each time I would "adjust Corners"  (make an adjustment) "done" "Adjust corner" make an adjustment  "done" etc...
Nothing would change, it was going back to SOFWARE ZERO, it wasn't checking the sensor each time....

My Workaround, or is this what you are supposed to be doing and no one told me....

To Adjust the Bed CENTER, I used "Home Z"  (make an adjustment) "Home Z" (make an adjustment) "Home Z" (make an adjustment)   and this time I could hear the microswitch click each time... 

What would be NICE is to make easier to do what I was doing by make the Z-Home STICK after a button push, so you can do z-home, z-home, z-home, etc... without having to do any scrolling....  to go BACK would require a scroll.

There are LOTS of new settings in the Motion area, specifically to do with the Z-Axis and around the bed leveling.... what do they do?   Can we adjust the Z-Axis in software separate from the Home-Sensor?

I am VERY used to the SnapMaker and its Magnetic bed sensor, it gets to +0.5mm or so magnetically, and then checks the corners with a feeler and adjust the height in software... works well, but we don't have that sensor, only the micro-switch and manual corner wheels.

Don't know if there can be something for BED TEMPERATURE in the bed leveling section, as it moves a LOT as it warms up, much more than the 0.13mm (what my calipers measured) feeler slip sheet.  It wouldn't be too hard to find the coefficient.  Set the bed COLD (25decC or so) to a known height and then raise it +50degC (makes math easy) or as hot as it will get, and re-measure the clearance and then figure out the change/degC.

Also need a "Pre-Heat BED" so you can Home the bed HOT or Cold....  I found out if you Home it COLD and then try to do back to back prints, the nozzle drags on the hot bed and does all sorts of bad things....

mikejr83, 2021-03-07:
Whoa. Lots of stuff here. Can you break out each of your requests to separate issues?

Is the largest concern that you want it to home z after it has done each of the corners?

1. Home Z
2. Center test
3. Front left
4. Front right
5. Back right
6. Back left
7. Home Z
8. Repeat...

The current process that I use is something similar but here:

1. Home Z
2. Level corners, start at center
3. Check the offset with the an index card
4. If too high or low repeat

Then do the level corners.

What you're suggesting is more like features. Please make sure you submit them as a feature so we can track them.

Oh, and thank you for your feedback!

inspectionsbybob, 2021-03-07:
Going to change these into FEATURE requests....

inspectionsbybob, 2021-03-07:
Not sure...
What I know is that "leveling the Bed" doesn't work well because the bed is
pivoted in the middle, sort of.
I just got everything so I could just feel the resistance... and then I
checked with a trusted straight edge... I had a dip in the middle from back
right, center to front left.  Back left to front right was OK...  But
everything was at 0.15mm... some something is screwy...
Bad Microswitch?  Sometimes I could hear the click, others not, so I think,
maybe, something is moving around...

So...

For this one, I will close this one and open 2-3 feature requests

Bob

closed 2021-03-07 by inspectionsbybob