JGMaker3dofficial/artistd #37: [BUG] Preheat PLA ALL not working

tjwitman, 2021-03-07:
### Bug Description

<When you select to preheat PLA ALL, it does not turn on the Bed>

### Mark's 2021.3-Beta release for JGMAKER ARTIST

**Required:** Files not changed from the release 
### Steps to Reproduce

1. Temperature
2. Preheat PLA
3.Preheat PLA ALL

**Expected behavior:** Expect the bed and both extruders to preheat

**Actual behavior:** only extruders turn on.  Bed stays set at 0

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mikejr83, 2021-03-10:
Verified this on 2.0.7.2 release.

mikejr83, 2021-03-10:
Turns out there was already a commit merged into upstream Marlin that fixed this issue.

mikejr83, 2021-03-11:
Issue is resolved in #45

closed 2021-03-11 by mikejr83