JGMaker3dofficial/artistd #38: [FR] Modify "Home-Z" to make adjusting the center easier.

inspectionsbybob, 2021-03-07:
Currently, to adjust the Center height you have to Home-Z, adjust, press, scroll to motion, scroll to Home Z, button, adjust, press, scroll to motion, scroll to Home Z, button, adjust, etc...

My suggestion is to change HOME-Z behavior to be

Back                                        **Home-Z** so the DEFAULT action is to "HOME-Z" and NOT RETURN to Home.

So Adjusting the Z axis would be press, scroll to motion, button, scroll to Home Z, button, button, adjust, button, adjust, button, adjust, etc....
and when DONE, scroll to back and button.

Would make adjusting (leveling) the Z-Axis MUCH easier...

inspectionsbybob, 2021-03-08:
I actually Released the Motors and adjusted the Bed far more quickly as I could move the Bed and the head R/L much faster by hand, and to where I wanted.   The "corners" alone doesn't cover all possibilities.   Right side (front and back at the same time), Front (R&L at the same time), and the other permutations actually got me to where I wanted quicker than using the "corners" routine.

This would require LOCKING-Z and releasing X&Y to prevent any slip (I didn't have any)

mikejr83, 2021-03-08:
FYI, the "Auto Home" items in the menus just run g-code. For instance, auto home z runs `G28Z`. This means it might be a bit of an effort to accomplish this. My thought is that adding it to the "level bed corners" area will be easier as it is actually backed by code instead of just a string of gcode to execute.

inspectionsbybob, 2021-03-08:
Most of How I am -currently- leveling is MANUAL, I don't even use the Corners routine.

What I do (currently) is:
HOME everyone
PLA Pre-Heat
Manually move the head to Middle  (or enter corner routine and exit after it moves to the middle)
unlock Motors
Z-Home   (adjust) (repeat)
Manually move to the right side, front, back, left, and adjust 2-thumb wheels at a time
Manually move to the corners and adjust 1-thumb wheel at a time
Double-check Middle
done
don't know how you can make this easier other than put all of the steps into one menu and make sure the things that need to be repeated don't force you back to the main info page each time.

GeraldHebinck, 2021-03-18:
One could use the #define CUSTOM_USER_MENUS in conf_adv to create a Levelingmenu with the following items (as macros).
Home -> G28
Center -> G1 Z5, G1 X155 Y155, G1 Z0
Home Z -> G28 Z, G1 Z0

just for adjusting the Z-Stop. 
and maybe even add the 4 corners.

I work with a home offset. My endstop triggers close to the bed and the last 0.2 i adjust via console.
https://marlinfw.org/docs/gcode/M206.html

mikejr83, 2021-03-18:
I've been using the custom menus on my personal build of the firmware for extruding 100mm of filament. This would be an excellent bit to add. I remember seeing some code around making it return back to the status screen after executing. If that could be prevented then this could make setting the z-limit much easier.

mikejr83, 2021-03-18:
Here's an idea I quickly came up with:
```c++
#define CUSTOM_USER_MENUS
#if ENABLED(CUSTOM_USER_MENUS)
  #define CUSTOM_USER_MENU_TITLE "JGMaker Commands"
  #define USER_SCRIPT_DONE "Operation Complete!"
  #define USER_SCRIPT_AUDIBLE_FEEDBACK
  //#define USER_SCRIPT_RETURN  // Return to status screen after a script

  // Used for z-limit adjustment. 
  // 1. Home all axis if necessary (O)
  // 2. Home only the z-axis every time
  // 3. Raises 5mm
  // 4. Go to the center of the bed
  // 5. Come down to 0 on the z-axis
  // When used back-to-back the homing is skipped for X and Y.
  // The z-axis will always home so that the adjustment will take effect.
  // The nozzle should just raise once for the homing and then 5mm up and back down.
  #define USER_DESC_1 "Z-Limit Adjustment"
  #define USER_GCODE_1 "G28 O\nG28 Z\nG0 Z5\nG0 X155 Y155\nG0 Z0"

  // Utility for calibrating e-steps for E0 (user sees E1)

  #define USER_DESC_2 "E-Steps Calibration E1"
  #define USER_GCODE_2 "G28 O\nM605 S1\nT0\nM109 S210\nG0 X155 Y155 Z50\nM83\nG1 F50\nG1 E100\nM104 S0"

  #define USER_DESC_3 "E-Steps Calibration E2"
  #define USER_GCODE_3 "G28 O\nM605 S1\nT1\nM109 S210\nG0 X155 Y155 Z50\nM83\nG1 F50\nG1 E100\nM104 S0"

  // #define USER_DESC_4 "Heat Bed/Home/Level"
  // #define USER_GCODE_4 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nG28\nG29"

  // #define USER_DESC_5 "Home & Info"
  // #define USER_GCODE_5 "G28\nM503"
#endif
```

inspectionsbybob, 2021-03-18:
Forgive me, I see this, and it makes sense, but have no idea how to
implement it, or if you are just sharing ideas on how it might be done.

My coding days are now very limited to very small projects I can code into
an Arduino in a few hours from scratch...  I don't have the knowledge to
pull in all the bit necessary to add this to the code base and compile a
custom image... sorry.

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

inspectionsbybob, 2021-03-18:
Ooops, sorry.  One thing I did see is that it assumes the bed is already
preheated...
If we are going to go to the trouble of coding this, we may want to make
sure the bed is pre-heated to at least PLA levels, or at least give a
warning that zeroing the Z-axis cold is not the best practice...

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

closed 2021-04-23 by github-actions[bot]