JGMaker3dofficial/artistd #39: [FR] Document all the settings added

inspectionsbybob, 2021-03-07:
The current release has a LOT of undocumented settings.

the request is to take some time and write a SHORT description of each setting in THREE PARTS.

1. Setting NAME as it appears on the screen
2. What it adjusts
3. What it would be used for

Example
Motion
- Bed Leveling
- - Fade Height ???
- - Bed Z ??

What are Fade Height and Bed Z ?

mikejr83, 2021-03-07:
All of these settings are stem from the base Marlin firmware.

Have you read through or tried searching on their site: https://marlinfw.org/docs/basics/introduction.html

I know the documentation isn't the best but I think some of what you're looking for is there.

We're considering putting together a firmware specific site and wiki. If you're up for helping (we love contributors!) coming up with a list of what needs to be documented, what is already covered in Marlin documentation, and helping come up with that documentation would be awesome!

inspectionsbybob, 2021-03-08:
i just spent a day chasing a non-problem. Bad filament was messing me up
but good.

I will go through the software in the next few days and see what I can do...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-03-08:
My First Pass will be an "Org-Chart" of the menu structure.  
It will have all of the menus and sub-menus on ONE easy-to-maintain document.  
What I have seen is a LOT of menu re-use that doesn't -quite- work.  Several routines call the "Change filament" routine, and show weird choices.  Others, like the PAUSE routine, call for filament change with no choice... and no way to exit.

closed 2021-05-02 by github-actions[bot]