JGMaker3dofficial/artistd #40: [FR] Automate Steps/mm calibration

inspectionsbybob, 2021-03-08:
Calibrating the Steps/mm is tough and requires lots of trial and error or MATH (frowny face)

If you asked people to mark on the filament 95, 100, 105, 110, and a few more and then ran 90mm and asked what was actually used, you could do the math and enter the correct value for them.

New value = current value * desired mm/actual mm

Repeat!

mikejr83, 2021-03-17:
We can do a custom menu where we tell the printer to extrude 100mm of filament at a certain temperature. This is pretty easy.

I'm not sure how far we're going to go down the rabbit hole of building features on top of stock Marlin. Even still, I think anything that would come close to completing this request would have to come after we've stabilized the firmware on 2.0.7.2.

inspectionsbybob, 2021-03-17:
We also need to fix the other extrusion issues such as idler wheel tension,
maybe a knurled wheel/gear, and other issues... no more washers... low
friction Run-out detector, etc. etc.

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

closed 2021-04-22 by github-actions[bot]