JGMaker3dofficial/artistd #41: [FR] Don't Block the Display with the bed

inspectionsbybob, 2021-03-08:
There are many steps where the Bed Moves to the full Forward location, Blocking the Display and the rotary selector

Add an option to NOT go full-forward, but to a position roughly 75% of forward so you can still see the menus

mikejr83, 2021-03-10:
I'm not sure how we would tackle this request. Can you please enumerate the scenarios where this occurs?

One quick thought, you may be seeing this because of your end gcode in your slicer. For instance, I have the printer "present" the print at the end of my prints. I do this by issuing a `G0` command and sending the bed all the way forward so it is easy to get the print off without having to work around the gantry. One important thing I do is unlock the steppers so that I can move things around freely by hand. If it's doing this in other occasions I'm not sure what we can do about it as it might be "as designed" by Marlin. We can try to accommodate the best we can, though.

inspectionsbybob, 2021-03-10:
Change filaments, or pause, is another time when it does a present...

it is confusing what is "Matlin" and what is "gcode"

i will start keeping track...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

mikejr83, 2021-03-10:
Ok. That's good to know. There may be configuration for that we can adjust. Thanks!

mikejr83, 2021-03-10:
This may be related to #28.

TwinkieXLII, 2021-03-16:
I know it's in the finishing moves and in M600

inspectionsbybob, 2021-03-16:
Question: In Beta 2  you said that saving to EEPROM had been fixed, did
they break it again??  That was only like 2 weeks ago ??

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

TwinkieXLII, 2021-04-23:
I want to verify this isn't adjustsble in the finishing code.

closed 2021-04-29 by github-actions[bot]