JGMaker3dofficial/artistd #42: [BUG] Don't assume the Bed is where you left it.

inspectionsbybob, 2021-03-08:
Any routine that is coming from a "motors unlocked" condition should assume that it DOESN'T KNOW where everything is and that things may have been moved.

Several times at the end of a run, I have moved the bed, or an extruder, for one reason or another, and when I continued on, the system ignored the limit switches and made lots of noise as it tried to move something to a location it was already at.

mikejr83, 2021-03-09:
Were you in duplication or mirror mode when you did this?

Most of the time marlin will make this assumption, that if you do certain actions that the machine is no longer homed. If you look at the display and see flashing numbers for X, Y, & Z then it should indicate that the machine will need to be homed before any other move.

I've noticed that when in duplication mode the machine does not respect limits correctly. This is the whole issue where if you do a `G28` after you've selected the duplication mode the machine tries to exceed its limits.

My thought is that we're uncovering issues in actual Marlin here and not necessarily in the JGMaker configuration. IDEX is still pretty new. Let's make sure we best document this so that we can raise it upstream with the Marlin team as well.

inspectionsbybob, 2021-03-09:
Nope, IDEX single extruder mode.  Haven't gotten to anything with the "other" head yet.

Still fighting weirdness... Now the Extruder is not heating up... Bed heats up, but as soon as it up to temp it tries to print, and gives a "M500 Error, Too Cold".  So In trying to tune the last print, I must have set something wrong...  It is sticking because it did the same thing when I reloaded the factory image...  Need to find out how to do a Wipe & reset (After I write down all the settings for travel and such)

mikejr83, 2021-03-10:
I think you might have an issue with your start gcode. Can you post that?

GeraldHebinck, 2021-03-18:
In the current configuration, the HOME_AFTER_DEACTIVATE is not enabled. Changing that will (most likely) fix the issue

closed 2021-04-23 by github-actions[bot]