JGMaker3dofficial/artistd #46: [BUG] power outage setting do not let print to start.

jpkuntz90, 2021-03-11:
Release version 2021.4-Beta

Printing do not start if
**Configuration** > **Power outage** setting is **on**.

Screen show directly :
> Power Outage : 
>    PRINTER HALTED
>    Please Reset"

TwinkieXLII, 2021-03-11:
Did you have an SD Card mounted at the time you started the print?

jpkuntz90, 2021-03-15:
Yes, i was ready to print.

mikejr83, 2021-03-15:
Thank you @jpflouret. May I just state the steps you ran through so that we know how to test this once we have a fix ready?

1. Slice model and save gcode to SD card
2. Insert SD card into printer
3. Check that power out resume is enabled in configuration
4. Go to "Print from media" menu
5. Select model
6. Start print
7. Error occurs as noted

I'll test this flow when my printer frees up from its next print.

jpflouret, 2021-03-15:
I think @mikejr83 tagged me incorrectly when the questions were meant for @jpkuntz90.

mikejr83, 2021-03-15:
Crap man sorry about that!

ale624, 2021-03-28:
Hi, 

I have just had this exact issue after updating to this latest firmware version.
The workflow is exactly as described above @mikejr83

TwinkieXLII, 2021-04-28:
I've got a fix that will be in 2021.3

closed 2021-05-04 by github-actions[bot]

ale624, 2021-05-04:
> I've got a fix that will be in 2021.3
> […](#)

Hey there. Did you mean 2021.5??

TwinkieXLII, 2021-05-04:
Yes 2021.5

Adricz22, 2025-08-14:
Hello @JGMaker3dofficial 👋

# 🌍 Top International Developer Events

Discover the **most influential tech conferences and developer events** happening around the world.  
Connect, learn, and grow with the **global developer community**.

---

## 🚀 Why Join?
- 🔴 **Live Events** – Stay updated with the latest happenings in tech  
- 🌐 **Global Community** – Network with developers from all continents  
- 📅 **Updated Daily** – Fresh events, always current  
- 🏆 **Top 10 Developer Events** – Hand-picked to shape the future of technology  

From **cutting-edge frameworks** to **emerging paradigms**, we bring you the opportunities that matter.

---

## 🔑 Get Started  
**[➡️ Continue with GitHub](https://www.globaldevsummit.com/)**  
🔒 **Connect securely** • Save favorites • Get personalized recommendations  

---

## 📌 Highlights
- Meet industry leaders and tech innovators  
- Join exclusive technical workshops  
- Explore new frameworks, tools, and best practices  
- Grow your personal and professional network  

---

> 📢 **Tip:** Bookmark the page and check daily for new opportunities.  
Your next big career move could start here!

---