JGMaker3dofficial/artistd #47: 2nd extruder move below 80mm position on x-axis when compiling from source

robertliou, 2021-03-15:
### When compiling from source code provided, unable to get second extruder to move below 80mm position on x axis.

I needed to compile my own code since I upgraded to hemera extruder x 2.  However, noticed when compiling from code unable to get extruder 2 to move below 80mm on x axis.  Extruder 1 is fine

### Steps to Reproduce

1. From source code, changed thermistors on E1 and E2
2. I changed the stepper driver to 2208 standalone for both E1&E2 (i upgraded the stock driver)
3. Inverted E2 direction so hemera will spin the correct direction

**Expected behavior:** [E2 should be able to move to 0,0 ]

**Actual behavior:** [E2 stops at 80,0 when asked to go 0,0]

#### Additional Information

mikejr83, 2021-03-15:
Ok, I thought this was me being unobservant, but I thought this was happening normally. Seeing this issue logged I decided to run `T1` and then `G0 X0 Y0`. It goes to 80,0 just as described.

I'll look into this more this evening.

mikejr83, 2021-03-16:
@robertliou I've created a pull request to fix this in the next official beta or release. If you want to apply the change so that you can compile it with your settings you can make the change to the following line: https://github.com/mikejr83/artistd/blob/feature/fix-x2-min-pos/Marlin/Configuration_adv.h#L625

robertliou, 2021-03-16:
I think i solved it.  Looking at configuration_adv x2 min position is set at 80.  It should be 3

TwinkieXLII, 2021-03-16:
I just put this fix into 2021.4.1 release

closed 2021-03-16 by TwinkieXLII