JGMaker3dofficial/artistd #49: Dual Color - E1 won't Home when E2 start to print (2021.4-RC1)

bodinjm, 2021-03-16:
### Description
ON 2021.4-RC1, When I launch a 2 color print, E1 won't Home when E2 start to print (2021.4-RC1)
I backup to 2021.4-beta and it does not make the Pb. 
probably the  #define X2_MIN_POS    80 => 5 change in last modification  

### Steps to Reproduce

Print the shift setting gcode makes the bug in last version and not in previous.  

**Expected behavior:** [What you expect to happen]

E1 should home when E2 start to print

**Actual behavior:** [What actually happens]
E1 stay in the middle of the plate when E2 came in then it crash into E1. 

#### Additional Information

mikejr83, 2021-03-16:
What IDEX mode are you running under? If you're using "Full control" then it will not move E1 back to its parked position. Use `M605 S1` to place the printer into auto park mode.

bodinjm, 2021-03-16:
Hello Mike,
Yes, I run full control default mode. What I don't understand is the need to switch to M605 S1 mode. It was not the case with 2021.4 version

mikejr83, 2021-03-16:
If you're running full control mode you need to tell the extruder to go park. In your slicer you'll need to tell each extruder to go to their parked location. This is how I understood it. the `M605 S1` command puts the printer into "Auto-park mode". When the firmware encounters a toolchange it will automatically send the previous extruder to its parked position.

Please double check your slicer settings if you're in full control mode.

closed 2021-04-22 by github-actions[bot]