JGMaker3dofficial/artistd #50: [BUG] (short description)

inspectionsbybob, 2021-03-16:
Odd one as I fight extrusion issues...

The system does not do a Z-Hop when it is going over the base layers....

My System is currently OVERextruding, which means I have little ridges in the Baselayers, and when the head travels from one side to the other to lay down another layer, I hear the BRrrrrrr of the extruder plowing through all the over extruded ridges.  Hasn't gotten to the in-fill or anything else yet (long print) but I re-started a few times thinking something was wrong or I was dragging through the bed only to figure out it was clipping the tops of the over extruded ridges....

With all the under extrusion problems, this is an oddball...  (I installed the Washer and took out the filament detector)

mikejr83, 2021-03-16:
Are you sure you're selecting "Z Hop When Retracted" otherwise I don't believe it does this in the firmware:

[image]

inspectionsbybob, 2021-03-16:
Yup, first thing I checked.  I don't think it considers the first (x)
layers to be wall, infill or tower.  It "hops" once it got past the
3-layers of the base, but not before...

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

mikejr83, 2021-03-17:
>  It "hops" once it got past the 3-layers of the base, but not before...

This may be a slicer issue. So far I do not think this is a firmware issue. Have you tried earlier versions of the firmware to see if it goes away?

inspectionsbybob, 2021-03-17:
No, I am trying to stay close to the release candidate so I don't
loose/forget settings that have been tweaked to get where I am now...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

TwinkieXLII, 2021-03-19:
Firmware zhop is disabled in rc1

inspectionsbybob, 2021-03-20:
Ah...
Had to roll back to Beta-4 anyway...  The current print is a 36hr monster... so it will be a while before I can play again...

inspectionsbybob, 2021-03-20:
Ah... The EEPROM is also disabled/not working.  Had to rollback to Beta-4
for other reasons as well.

Lost my Matrix last flash, and had to re-do that.  In the middle of a 40hr
print, so it will be a while until I can figure out if I was on RC1 or the
"other" non-factory release that was giving me fits...

Bob S.

closed 2021-05-11 by github-actions[bot]