JGMaker3dofficial/artistd #52: [BUG] Home {axis} is missing

inspectionsbybob, 2021-03-19:
Ok, just flashed Release candidate-1
Love the Extruder dance over the brush
However

- It calls for a filament change at the beginning of a print no matter what
- **Home X, Home Y and Home Z commands are MISSING**
   (That makes it Very hard to Zero/Level the Z-axis)
- The bed ZOOMS forward at the end of an abort (ouch) blocking the panel 
      (can we do 75% so we can still see the panel)
    
[image]

-

mikejr83, 2021-03-19:
Are you sure you're using the firmware from this release and not my (from my GitHub account) release? I just double checked and unless the wrong bin was uploaded the version that should be out there has individual axis homing enabled.

inspectionsbybob, 2021-03-19:
I will double-check, could have sworn it was RC-1...
Sometimes confusing, sorry if it wasn't yours... I will check when my print
is done (I rolled back to Beta-4)

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

closed 2021-04-14 by inspectionsbybob