JGMaker3dofficial/artistd #58: E2 Crash e1 while man Filament

ThiloSchneider, 2021-04-05:
### Bug Description

Hi Michael Gardner...

Found an issue while changing Filament via fw 
 e2 
Rod Off
E1 inactiv
When i start the Filament Change e2 while printing it runs to the left and Crashes Into e1 and the Stepper Stalls and loose its Position. 
When the Print Continues  Pr the Position is Further right Then before the change

### Last GitHub fw

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Change Fil while printing with E2, e1 inactive
2. Choose by the knob 

**Expected behavior:** z Hop and Home e2

**Actual behavior:** [What actually happens]
Z Hop and Crash in Parker e1 

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

daywalkerbe, 2021-04-22:
I have the same problem with the Last firmware

closed 2021-05-28 by github-actions[bot]

TwinkieXLII, 2021-09-28:
THis is still an issue.  I think it can be fixed with HOME_BEFORE_FILAMENT_CHANGE

nickm324, 2023-06-09:
This is still an issue. It has been two years, any chance this is ever going to get fixed. I cannot believe that JGMaker has abandoned this printer already and hasn't released an updated firmware in 2 years.