JGMaker3dofficial/artistd #64: [BUG] cannot flash mks robin pro with self compiled source

tnap1979, 2021-07-17:
### Discussed in https://github.com/JGMaker3dofficial/artistd/discussions/63

<div type='discussions-op-text'>

<sup>Originally posted by **tnap1979** July 16, 2021</sup>
so im bloody lost....
im far from new at board flashing and compiling, but this board has me stumped...
its the only one that refuses to finish flashing self compiled stuff... but if i take a pre-compiled bin from here or jgmaker site, it works fine and flashes and boots...
otherwise, every firmware ive tried to compile myself whether stock source, or modified source or whatever...
it ALWAYS hangs at 99% and nevers finishes... always left with a display that says " boot " on every power up..
but flashing pre-compiled latest release bin from here... i get the jgmaker logo and animated marlin boot and it works..
my intentions are to compile my own so i can add bltouch and tmc-2209's and some other stuff id rather modify myself then have someone else do for me..
my vscode has been updated, all platformio modules updated..</div>

mikejr83, 2021-07-17:
What's the name of the file that you're flashing from the build? Have tried different sd cards?

rclarke250, 2021-07-17:
Are you putting the assets folder along with the robin_pro35.bin file on the sd card? the touch screen does not have it's own FW, it relies on the MCU of the Robin pro board to supply that. For FW older than Marlin 2.07 like the OEM FW which is based on Marlin 2.05.03 it uses older LVGL version, that is not compatible with the newer versions of Marlin, which uses LVGL 7. If you use the older version you have to include the /fonts and /pics folders from the OEM FW. I am sure there are settings that allow it to not erase that portion of the eprom when flashing the firmware, like adding an offset to start the flash, but I do not know it, and because MKS github for the robin_pro and robin_tft35 is crap, and never been updated with actual helpful information, we might not ever know, I decided to just yank the board out, and go with what works, I asked MKS to update the Github, and the online configuration tool, months ago, and they never have. The last I had working was Marlin 2.08 but it was missing the mode buttons for changing X axis modes.

closed 2021-08-23 by github-actions[bot]