JGMaker3dofficial/artistd #66: [FR] How to add a bltouch

Gazgiza, 2024-11-17:
Hi ! I am having trouble finding the file to modify to add the bltouch option for my printer like shown in this video : [https://www.youtube.com/watch?v=vmxOyTHsbhQ](url)

Where can I find the original file that I can modify for this purpose ?

Thanks
Have a nice day

ellensp, 2024-11-19:
@Gazgiza please close this one too

Gazgiza, 2024-11-19:
[https://github.com/JGMaker3dofficial/artistd/issues/67](url)

Here is the fix

Gazgiza, 2024-11-20:
Hi! I am now having trouble with the bed auto leveling... It is only probing one point instead of the mesh that is supposed to be probed. 
I've set it to be a grid of 6x6 :
```
  #define GRID_MAX_POINTS_X 6
  #define GRID_MAX_POINTS_Y 6
```

Gazgiza, 2024-11-20:
[Configuration.zip](https://github.com/user-attachments/files/17827735/Configuration.zip)

closed 2024-11-20 by Gazgiza