JGMaker3dofficial/artistd #67: [BUG] When trying to compile a custom firmware

Gazgiza, 2024-11-17:
Hi ! I'm trying to compile a custom firmware for my jgmaker artist d using the marlin firmware auto build extension for vs code but, once I hit compile, I get this error :

``` Warning!` Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F7] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
NameError: name 'env' is not defined:
  File "C:\Users\youri\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 180:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "C:\Users\youri\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\youri\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\youri\Downloads\artistd-main\buildroot\share\PlatformIO\scripts\mks_robin_pro.py", line 6:
    for define in env['CPPDEFINES']:
=============================================================================== [FAILED] Took 4.28 seconds ===============================================================================

Environment    Status    Duration
-------------  --------  ------------
mks_robin_pro  FAILED    00:00:04.282 
```
As I am new to this extension, I've looked on the intrenet trying to find what was wrong but nothing seems to match with my issue... Even chat gpt doesn't know...

ellensp, 2024-11-17:
from https://github.com/JGMaker3dofficial/artistd/blob/main/README.md

# Firmware For JGMaker Artist-D
This firmware is no longer supported by the community.  

Please use the Twinkie Fork.   
* Artist-D Pro [https://github.com/TwinkieXLII/ArtistD-Pro](https://github.com/TwinkieXLII/ArtistD-Pro)
* Artist-D Standard [https://github.com/TwinkieXLII/artistd](https://github.com/TwinkieXLII/artistd)

ellensp, 2024-11-18:
Of course that doesn't work either, and they have turned off issues 

Use https://github.com/TwinkieXLII/artistd

apply this diff to buildroot/share/PlatformIO/scripts/mks_robin_pro.py

```DIFF
diff --git a/buildroot/share/PlatformIO/scripts/mks_robin_pro.py b/buildroot/share/PlatformIO/scripts/mks_robin_pro.py
index 2ebf1ffb9..8054de810 100644
--- a/buildroot/share/PlatformIO/scripts/mks_robin_pro.py
+++ b/buildroot/share/PlatformIO/scripts/mks_robin_pro.py
@@ -2,7 +2,8 @@ import os
 Import("env")
 
 # Relocate firmware from 0x08000000 to 0x08007000
-for define in env['CPPDEFINES']:
+envdefs = env['CPPDEFINES'].copy()
+for define in envdefs:
     if define[0] == "VECT_TAB_ADDR":
         env['CPPDEFINES'].remove(define)
 env['CPPDEFINES'].append(("VECT_TAB_ADDR", "0x08007000"))
```

or just replace it with the mks_robin_pro.py in this zip file
[mks_robin_pro.zip](https://github.com/user-attachments/files/17792991/mks_robin_pro.zip)

This still has warnings about src_filter so will break sometime in the future when they remove it.
But for now it builds.

ellensp, 2024-11-18:
for a full patch that resolves all platformio warnings see the PR

Gazgiza, 2024-11-18:
Hi ! Thank you for your answer. It helped me a bit. The compilation seems to have gone a step forward but I now get this error coming from the probing thechnique I chose. It is asking me to chose only one of the option, I have chosen the one with bltouch commented on the same line and have disabled the one that was on by default.

Here is the log :

`
PS C:\Users\youri\Downloads\artistd-Twinkie-2022.2.2> platformio run --silent -e mks_robin_pro
Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Tool Manager: Installing platformio/framework-arduinoststm32 @ ~4.10900.0
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoststm32@4.10900.200819 has been installed!
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:30:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1513:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, MAGLEV4, or Z Servo."
 1513 |   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, MAGLEV4, or Z Servo."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1513:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, MAGLEV4, or Z Servo."
 1513 |   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, MAGLEV4, or Z Servo."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1513:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, MAGLEV4, or Z Servo."
 1513 |   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, MAGLEV4, or Z Servo."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1513:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, MAGLEV4, or Z Servo."
 1513 |   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, MAGLEV4, or Z Servo."
      |    ^~~~~
[... cut ...]
      |    ^~~~~
*** [.pio\build\mks_robin_pro\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o] Error 1
*** [.pio\build\mks_robin_pro\src\src\HAL\shared\eeprom_if_spi.cpp.o] Error 1
*** [.pio\build\mks_robin_pro\src\src\HAL\shared\esp_wifi.cpp.o] Error 1
*** [.pio\build\mks_robin_pro\src\src\HAL\shared\servo.cpp.o] Error 1
*** [.pio\build\mks_robin_pro\src\src\MarlinCore.cpp.o] Error 1
`

And here is the updated file : 
[Configuration.zip](https://github.com/user-attachments/files/17804688/Configuration.zip)

ellensp, 2024-11-18:
You cannot have
#define PROBE_MANUALLY and #define BLTOUCH  ie two different probes, just as the error said.

Gazgiza, 2024-11-19:
Thank you very much. It has now compiled. 
Thanks you for your help.

Gazgiza, 2024-11-20:
Hi! I am now having trouble with the bed auto leveling... It is only probing one point instead of the mesh that is supposed to be probed.
I've set it to be a grid of 6x6 :

```
  #define GRID_MAX_POINTS_X 6
  #define GRID_MAX_POINTS_Y 6
```

How may I make so that it probes all the points ?
[Configuration.zip](https://github.com/user-attachments/files/17831918/Configuration.zip)

closed 2024-11-21 by Gazgiza